ajn96/iSensor-SPI-Buffer #4: Add SD card data logging capability

ajn96, 2020-06-26:
The iSensor-SPI-Buffer hardware platform has a SD card slot connected to SPI3. Need to develop software support for autonomous data logging to the SD card (for use in battery powered applications without a PC).

juchong, 2020-07-24:
In an effort to prevent data corruption, we might be able to use the [LTC4412](https://www.analog.com/media/en/technical-documentation/data-sheets/4412fb.pdf) to detect power loss and swap over the MCU supply to a supercap backup. The proposed supercap is [CPH3225A-2K](https://www.digikey.com/product-detail/en/seiko-instruments/CPH3225A-2K/728-1067-1-ND/4747400).

ajn96, 2020-08-15:
Software support in progress. Should be mostly done, having issue getting SD card to mount correctly.

ajn96, 2020-10-01:
Finally done!!! Want to resolve USB data corruption issues, then will push a firmware rev 2.0 release with this data logging feature

closed 2020-10-01 by ajn96