ajn96/iSensor-SPI-Buffer #10: Loss of PPS input question

NevadaMark, 2020-07-27:
In the second figure, under the TIMESTAMP_UPR register description, it appears like the period of the last PPS signal is shorter than the previous two pulses, as evidenced by attributes in the UTC_TIME_LWR (time period of the last increment was shorter, when compared to the previous two cycles.  
https://github.com/ajn96/iSensor-SPI-Buffer/blob/master/REGISTER_DEFINITION.md#TIMESTAMP_UPR

Could this be indicative of a signal artifact, which came from the act of removing the PPS clock signal?  In other words, did a glitch cause this to trigger sooner than it would have otherwise or is there another reason for the difference in period?  Thank you! 

[image]

ajn96, 2020-07-27:
Hi Mark,

You're correct, the short period on the last PPS input period was likely due to an input glitch when disabling the PPS signal. 

The PPS input to the SPI buffer for this image was being generated by an FX3 PWM output (1Hz, 50% duty cycle). When a PWM output is disabled on the FX3, the pin is tri-stated. This caused a low to high transition to be seen by the SPI buffer, which triggered the PPS ISR (DIO_INPUT_CONFIG PPS Polarity bit was set to 0).

Do you think it would be valuable to detect an error like this for the PPS_UNLOCK bit? Currently, this bit is only set when the PPS signal is lost - it would be easy enough to flag an invalid PPS period as well (allowed freq of 1Hz +- 3%).

NevadaMark, 2020-07-27:
Thank you, Alex!
This confirmation helps strengthen my understanding of this function, which by the way, is pretty cool!  
With respect to detecting an error, based on the time period, I can imagine that being useful.  What would limit the precision of this detection method?

ajn96, 2020-07-27:
> Thank you, Alex!
> This confirmation helps strengthen my understanding of this function, which by the way, is pretty cool!
> With respect to detecting an error, based on the time period, I can imagine that being useful. What would limit the precision of this detection method?

There are two major limits to the precision of the PPS error detection: 
* CPU activity can make interrupt latency (to process PPS edge) somewhat variable. The PPS input has a lower interrupt priority than the user SPI, so if the processor is executing the user SPI ISR when the PPS input comes in the "measured" period could be off by ~1us. In addition, the PPS input shares the same interrupt vector as the IMU data ready, so if an IMU data ready comes in just before the PPS that could push the inaccuracy of the period measurement out further.
* Accuracy of the crystal used to clock the SPI buffer. 20ppm shouldn't drift much over 1 second, but wouldn't want to pull the margins in too close and start reporting false PPS_UNLOCK failures.

Thinking about the possible error sources more now, I suspect I could very safely assert an error if the PPS period is off by +-10ms (1%).

It might be possible to set up another timer peripheral on the ST processor to operate in input capture mode to monitor the PPS input, which would give much more precise and accurate timing measurements.

NevadaMark, 2020-07-28:
Thank you!  Based on what I know so far, +/-1% seems like it would suffice.  Perhaps some applications will value more precise detection, but I am not sure that should be the responsibility of the IMU (which in this context, would include this FIFO/Buffer). I will keep thinking about this, in my research.  Thank you for the education on the capability.

ajn96, 2020-07-28:
Added +- 1% error detection on release 1.3

closed 2020-07-28 by ajn96