ajn96/iSensor-SPI-Buffer #9: TIMESTAMP_UPR negative value?

NevadaMark, 2020-07-27:
Should the minimum value of the TIMESTAMP_UPR register be zero or is this a twos complement number?  Thank you! 
https://github.com/ajn96/iSensor-SPI-Buffer/blob/master/REGISTER_DEFINITION.md#TIMESTAMP_UPR

[image]

ajn96, 2020-07-27:
Hi Mark,

The timestamp register is unsigned, 32-bits. The negative is due to an offset applied by the FX3 data plotting GUI by the "autonull" functionality. I had autonulled the plot to make it easier to see both signals.

I'll generate a new plot with the offset set to 0 for the timestamp, to avoid potential confusion.

ajn96, 2020-07-28:
Updated image

closed 2020-07-28 by ajn96