ajn96/iSensor-SPI-Buffer #11: TIMESTAMP_UPR Peak Value, when using 1Hz PPS Signal

NevadaMark, 2020-07-28:
If I understand correctly, and TIMESTAMP_UPR and TIMESTAMP_LWR represent a 32-bit counter, which represents the number of microseconds (1us = 0.000001s) since the trigger edge of the PPS signal.   Assuming ideal conditions, that will result in the following value for the TIMESTAMP:

- TIMESTAMP = 0x000F4240 (hexadecimal equivalent of 1000000)
- TIMESTAMP_UPR = 0x000F
- TIMESTAMP_LWR = 0x4240

If TIMESTAMP_UPR should be 0x000F, which represents a decimal of 15, why is the peak value, in this figure, only 14?  
[image]

ajn96, 2020-07-28:
Hi Mark,

This cap at 14 is likely an artifact due to the relatively low sample frequency the plotting for was set to (20Hz). Reading the timestamp registers faster shows results in-line with what we would expect:

[image]

Running the same plot again with a higher Fs sometimes "catches" the timestamp upper register at 15
[image]

NevadaMark, 2020-07-28:
Thank you, Alex!
That makes sense. Thank you for running that experiment.  
If we divide 10^6 by 65535 (2^16 - 1), we arrive at a nominal value of 15.259.  
If we subtract 15 from this number, we arrive at 0.259 is ~1.7% of 15.259 (nominal expectation), which is not far from the +/-1% tolerance that you predicted for the tolerance in detecting clock frequency error (in a separate issue).  Could the combination of accuracy of the PPS signal and accuracy of the measurement method also be contributing to the variation in the count, for each cycle?  Again, many thanks for supporting the discussion!  This is good stuff!

ajn96, 2020-07-29:
Hi Mark,

Sorry for the misunderstanding - the TIMESTAMP_UPR register in the plot above is truly just the upper 16 bits of the microsecond timestamp. The lower 16 bits were not plotted individually because it changes too fast (0 - 65535 every time TIMESTAMP_UPR increments). I guess I should probably set one of the registers up as 32-bit in the regmap!

The measurement variation is just a function of the "slow" plotting utility. For each PPS period, the TIMESTAMP_UPR is only at 15 for ~16ms (0x4240 microseconds) before being reset to 0. The plotter utility only reads the register every 50ms with Fs = 20Hz, so the odds of it "catching" the TIMESTAMP_UPR register at 15 aren't great.

NevadaMark, 2020-07-29:
Hi Alex,
This makes sense to me! Thank you for offering this level of detail and explanation. 
Mark

closed 2020-07-31 by ajn96