ajn96/iSensor-SPI-Buffer #17: Add additional decoupling for the IMU

juchong, 2020-08-17:
Resetting the IMU causes VDD to fluctuate significantly. More decoupling should be added to lessen this behavior. 
[image]

ajn96, 2020-08-24:
Added a decimation filter to the temp sensor output to make it look a little cleaner. Can still see the noise from a messy supply being multiplied into the temp measurement, but it's not as bad.
[image]

ajn96, 2021-04-06:
This has been incorporated into the rev C board design (100 puf caps decoupling IMU supply)