ajn96/iSensor-SPI-Buffer #18: Add a solder pad that disconnects the 24-pin connector VDD input from the buffer board

juchong, 2020-08-31:
This is useful to avoid back-driving 3.3V into a pre-existing system.

ajn96, 2021-04-06:
@juchong On the marked up schematic for rev C of the buffer board, you added a solder jumper for VUSB vs VDD as the input to the regulator. Did you mean to have VUSB vs VIN? Or is this issue referencing a different pad?
[image]

ajn96, 2021-07-22:
Adding two solder jumpers:

1. First allows selection of input to the regulator from VDD_USB or VIN
2. Second allows selection of IMU supply voltage as 3.3 or VIN

This allows a higher voltage IMU to operate from VDD_USB or VIN directly, while the STM32 operates from a 3.3V regulated supply with minimal rework required