ajn96/iSensor-SPI-Buffer #28: Use the unallocated pin on the USBC connector as a PPS input pin

juchong, 2020-10-22:


ajn96, 2020-10-23:
@juchong good idea! We should be able to use on of the dedicated USB alternate function pins. I'll need to take a look at what pin would be good on the micro. In general before you start the layout on the Rev C we should probably do a full schematic review

juchong, 2020-11-09:
An additional adapter/injection board will need to be developed to take advantage of this feature.

ajn96, 2021-04-06:
This has been incorporated into the rev C board design. The USB alternate function line goes into a mux, which can be configured by software to connect the line to any of the IMU DIO lines.

ajn96, 2021-11-03:
Plan is to add a "SB_MUX" bitfield to CLI_CONFIG to support this mux functionality